microsoft/mu_tiano_platforms #111: GCC Compiled Q35 Faults in SMM when using KVM acceleration

cfernald, 2022-10-27:
The device hits a general protection fault when booting with KVM acceleration on a GCC compiled SMM supervisor. Currently, the EXDI debugger causes a QEMU assertion when connecting in the SMM supervisor and so debugging has been difficult. Confirmed that the Visual Studio compiled supervisor does not hit this exception when running in KVM acceleration. 

`
INFO - Loading MM driver at 0x0007FD28000 EntryPoint=0x0007FD2C0B0 MmSupervisorRing3Broker.efi
INFO - StartImage - 0x7FD2C0B0 (Standalone Mode)
INFO - !!!! X64 Exception Type - 0D(#GP - General Protection)  CPU Apic ID - 00000000 !!!!
INFO - ExceptionData - 0000000000000000
INFO - RIP  - 000000007FFDBAB8, CS  - 0000000000000038, RFLAGS - 0000000000010006
INFO - RAX  - 000000000000002F, RCX - 000000007FEC273E, RDX - 0000000000000001
INFO - RBX  - 000000007FEEDF60, RSP - 000000007FEC26E8, RBP - 000000007FFFB1D0
INFO - RSI  - 000000007FD95FC0, RDI - 000000007FEC2906
INFO - R8   - 0000000000000000, R9  - 000000007FD59FFF, R10 - 0000000000000000
INFO - R11  - 0000000000000000, R12 - 000000007FFEF3D4, R13 - 000000007FEEDF78
INFO - R14  - 000000007FFF30E0, R15 - 000000007FEBBFF8
INFO - DS   - 0000000000000040, ES  - 0000000000000040, FS  - 0000000000000040
INFO - GS   - 0000000000000040, SS  - 0000000000000040
INFO - CR0  - 0000000080010033, CR2 - 0000000000000000, CR3 - 000000007FE22000
INFO - CR4  - 0000000000100E68, CR8 - 0000000000000000
INFO - DR0  - 0000000000000000, DR1 - 0000000000000000, DR2 - 0000000000000000
INFO - DR3  - 0000000000000000, DR6 - 00000000FFFF0FF0, DR7 - 0000000000000400
INFO - GDTR - 000000007FDA8000 000000000000007F, LDTR - 0000000000000000
INFO - IDTR - 000000007FEB4000 00000000000001FF,   TR - 0000000000000070
INFO - FXSAVE_STATE - 000000007FEC2340
INFO - SmmSupervisorMiscExceptionHandler MM Supervisor fault here
INFO - It is invoked from the instruction before IP(0x7FFDBAB8) in module (/workspaces/mu_tiano_platforms/Build/QemuQ35Pkg/DEBUG_GCC5/X64/MmSupervisorPkg/Core/MmSupervisorCore/DEBUG/MmSupervisorCore.dll)
INFO - SmmSupervisorMiscExceptionHandler MM Supervisor error reporting failed - Not Ready!!!
INFO - SmmSupervisorMiscExceptionHandler - Reboot here in test mode.
`

makubacki, 2023-05-19:
@cfernald - come back and test this

makubacki, 2023-06-01:
@cfernald - Check this

cfernald, 2023-06-06:
@kuqin12 FYI

This does not appear to hit on the newest tools & code combination. Closing the issue.

closed 2023-06-06 by cfernald