microsoft/mu_tiano_platforms #240: [Feature]: Enable Q35 to run without SMM

cfernald, 2023-01-20:
### Feature Overview

Q35 currently requires SMM to execute, but this limits the ability to run under the Windows HVX acceleration. Q35 should be able to boot, fully functional, without SMM being supported by the machine. 

### Solution Overview

Create a build flag that will swap out the use of SMM and standalone MM for the DXE/runtime equivalents for flash access, etc. 

### Urgency

Medium

### Are you going to implement the feature request?

I will implement the feature

makubacki, 2023-04-24:
@cfernald this can be closed now, right?

cfernald, 2023-04-24:
Yep

closed 2023-04-24 by cfernald