microsoft/mu_tiano_platforms #522: [Feature]: Enable Advanced Logger in PEI and MM on Q35

TaylorBeebe, 2023-04-26:
### Feature Overview

Advanced Logger is enabled in DXE on Q35 and should be enabled in all phases.

### Solution Overview

Add PEIM, PEI_CORE, MM_CORE, and MM_STANDALONE instances of AdvancedLoggerLib and the Advanced Logger DebugLib.

### Urgency

Low

### Are you going to implement the feature request?

I will implement the feature

### Do you need maintainer feedback?

No maintainer feedback needed

TaylorBeebe, 2023-12-19:
The following PR fulfills this request: https://github.com/microsoft/mu_tiano_platforms/pull/801

kuqin12, 2023-12-20:
> The following PR fulfills this request: #801

The #801 will not be able to cover MM_CORE and MM_STANDALONE due to the fact that current advanced logger implementation is relying on the memory buffer to be initialized from TFA, which will require further customization.

The modification of TFA for broader usage needs separate planning. We will discuss this internally before any public movement here.

PR #801 (2023-12-20): Add Advanced logger for Non DXE phases
## Description

This change introduces non DXE advanced logger, which should enable us to verify the advanced logger related changes better. Resolves #522.

- [x] Impacts functionality?
  - **Functionality** - Does the change ultimately impact how firmware functions?
  - Examples: Add a new library, publish a new PPI, update an algorithm, ...
- [ ] Impacts security?
  - **Security** - Does the change have a direct security impact on an application,
    flow, or firmware?
  - Examples: Crypto algorithm change, buffer overflow fix, parameter
    validation  [... cut]
commit c2982a82 (2023-12-20): Add Advanced logger for Non DXE phases (#801)

## Description

This change introduces non DXE advanced logger, which should enable us
to verify the advanced logger related changes better. Resolves #522.

- [x] Impacts functionality?
- **Functionality** - Does the change ultimately impact how firmware
functions?
- Examples: Add a new library, publish a new PPI, update an algorithm,
...
- [ ] Impacts security?
- **Security** - Does the change have a direct security impact on an
application,
    flow, or firmware?
  - Examples: Crypto algorithm change, buffer overflow fix, parameter
    validatio [... cut]
closed 2023-12-20 by kuqin12