microsoft/mu_tiano_platforms #1114: [Feature]: Consider a Release Branch and Dev Branch Test Strategy

makubacki, 2025-02-12:
### Feature Overview

Project Mu recently moved repos with `release` branches to a `dev` branch by default. This feature request is to determine whether to build both the `release` and `dev` branches in CI.

To align with the default branch in other repos and make new changes and features more readily accessible, https://github.com/microsoft/mu_tiano_platforms/pull/1112 moved the default branches used in Mu Tiano Platforms Project Mu submodules to dev branches.

### Solution Overview

Since Mu consumers may use either a `dev` or `release` branch in a platform, this feature tracks consideration of testing both in Mu Tiano Platforms. This may be challenging due to backward incompatible changes made in the dev branch and common platform code being used across both. But ways to address that can be discussed further here.

### Urgency

Low

### Are you going to implement the feature request?

Someone else needs to implement the feature

### Do you need maintainer feedback?

Maintainer feedback requested

os-d, 2025-02-12:
I agree with mu_tiano_platforms using the `dev` branches for regular development, it is a development vehicle in the end.

For testing release branches with it, because we don't care as much if a platform level change breaks the release branch (that would just break mu_tiano_platforms, not other consumers of the release branch), I wonder if we want to just run a nightly CI (or some cadence) with the release branches to ensure that the integration of the branches is correct (which our consumers do need). To me this makes more sense than a CI gate, because it would block mu_tiano_platforms PRs for something that likely would need to be fixed in the submodule, due to some incompatibilities between repos.

This does lead to failing nightly CI not being a blocking issue, so would it get enough attention to fix, but I think we can prioritize it because the release branches affect all of our consumers.

The other challenge will be dev and release branch divergence: if mu_tiano_platforms takes a dependence on a feature only in dev or some breaking change, it would have to be gated to also support release and then on the next integration would need to be removed, so this is additional overhead.

makubacki, 2025-02-12:
Thanks Oliver. We discussed the following solution offline that incorporates some of the elements there.

1. Have a `main` branch (mu_tiano_platforms default branch) that points to the latest `dev` branches.
2. Have a `release` branch that points to the latest `release` branches.

This allows:

- Clear tracking of submodule state.
- Simple understanding of each branch state.
- Ability to bind platform changes against upstream branches independently with no conditional or special CI logic.

---

This also simplifies PR gates and CI as a PR into each target branch can trigger the respective checks for that branch.

makubacki, 2025-02-12:
Assigned to @apop5.

apop5, 2025-04-01:
Not stale.

Will happen with 202502 branches.

apop5, 2025-05-17:
Not stale.

Linked #1142 as first step.

makubacki, 2025-06-08:
- @apop5, 1142 moved the default for 2502 to dev branches, but that was already decided upon (as noted in  this issue description).
- This tracks testing both dev and release branches, which doesn't appear to be done. Is that correct?
  > Since Mu consumers may use either a dev or release branch in a platform, this feature tracks consideration of testing both in Mu Tiano Platforms.

makubacki, 2025-09-15:
This issue is no longer relevant. Project Mu has decided to return to a single branch and drop the dev/release branch split. Closing the issue.

closed 2025-09-15 by makubacki