microsoft/mu_tiano_platforms #1148: [Bug]: Error when trying to build QemuSbsaPkg

edhay, 2025-05-21:
### Is there an existing issue for this?

- [x] I have searched existing issues

### Current Behavior

When trying to build QemuSbsaPkg, stuart_update fails:

```
$ stuart_update -c Platforms/QemuSbsaPkg/PlatformBuild.py TOOL_CHAIN_TAG=GCC5
SECTION - Init Self Describing Environment
SECTION - Loading Plugins
SECTION - Start Invocable Tool
SECTION - Initial update of environment
UpdatingWARNING - [SDE] Failed to fetch NugetDependecy: mu-uncrustify-release@73.0.8: [Nuget] We failed to install this version 73.0.8 of mu-uncrustify-release
WARNING - [SDE] Failed to fetch NugetDependecy: mu-disc-utils-bin@0.0.1: [Nuget] We failed to install this version 0.0.1 of mu-disc-utils-bin
WARNING - [SDE] Failed to fetch NugetDependecy: edk2-basecrypto-driver-bin@1.0.1: [Nuget] We failed to install this version 1.0.1 of edk2-basecrypto-driver-bin
WARNING - [SDE] Failed to fetch NugetDependecy: mu_nasm@20016.1.1: [Nuget] We failed to install this version 20016.1.1 of mu_nasm
WARNING - [SDE] Failed to fetch NugetDependecy: edk2-acpica-iasl@20230628.0.1: [Nuget] We failed to install this version 20230628.0.1 of edk2-acpica-iasl
. Done
SECTION -       Updated/Verified 2 dependencies
SECTION - Second pass update of environment
UpdatingWARNING - [SDE] Failed to fetch NugetDependecy: mu-uncrustify-release@73.0.8: [Nuget] We failed to install this version 73.0.8 of mu-uncrustify-release
WARNING - [SDE] Failed to fetch NugetDependecy: mu_nasm@20016.1.1: [Nuget] We failed to install this version 20016.1.1 of mu_nasm
WARNING - [SDE] Failed to fetch NugetDependecy: edk2-basecrypto-driver-bin@1.0.1: [Nuget] We failed to install this version 1.0.1 of edk2-basecrypto-driver-bin
WARNING - [SDE] Failed to fetch NugetDependecy: mu-disc-utils-bin@0.0.1: [Nuget] We failed to install this version 0.0.1 of mu-disc-utils-bin
WARNING - [SDE] Failed to fetch NugetDependecy: edk2-acpica-iasl@20230628.0.1: [Nuget] We failed to install this version 20230628.0.1 of edk2-acpica-iasl
. Done
SECTION -       Updated/Verified 2 dependencies
ERROR - We were unable to successfully update 5 dependencies in environment
SECTION - Summary
ERROR - Error
```

 Consequently the stuart_build too fails. The error log is attached. 

[QemuSbsa build error.txt](https://github.com/user-attachments/files/20361550/QemuSbsa.build.error.txt)

### Expected Behavior

Build should succeed. 

### Steps To Reproduce

Follow the steps at https://github.com/microsoft/mu_tiano_platforms/blob/main/Platforms/Docs/Common/building.md

### Build Environment

```markdown
- OS(s):Ubuntu 22.04.01
- Tool Chain(s): GCC5
- Targets Impacted: AARCH64
```

### Version Information

```text
Commit: dfeaac1ea99b8b806b3d0f0f6a41c34dd0c608ba
```

### Urgency

Medium

### Are you going to fix this?

Someone else needs to fix it

### Do you need maintainer feedback?

Maintainer feedback requested

### Anything else?

None

apop5, 2025-05-27:
@edhay 

I"m assuming that you are doing this under a linux environment? Or WSL?

All the failued updates appear to be coming from NUGET.

Can you verify that you installed mono-complete in the distro that you are using?

edhay, 2025-06-05:
Hi @apop5
Apologies for the delayed replied.
I am building on Linux env.
`Linux a076513 6.2.0-39-generic #40~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Nov 16 10:53:04 UTC 2 x86_64 x86_64 x86_64 GNU/Linux`

I found that `mono-complete` was not installed. Now installed the same and recompiled.
Now the error message has changed:

```
INFO - Building ... /home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/MU_BASECORE/MdePkg/Library/StackCheckLib/StackCheckLibDynamicInit.inf [AARCH64]
INFO - Trim --asm-file -o /home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/Build/QemuSbsaPkg/DEBUG_GCC5/AARCH64/MdePkg/Library/StackCheckLib/StackCheckLibDynamicInit/OUTPUT/AArch64/StackCookieInterrupt.i -i /home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/Build/QemuSbsaPkg/DEBUG_GCC5/AARCH64/MdePkg/Library/StackCheckLib/StackCheckLibDynamicInit/OUTPUT/inc.lst /home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/MU_BASECORE/MdePkg/Library/StackCheckLib/AArch64/StackCookieInterrupt.S
INFO - Building ... /home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/MU_BASECORE/MdePkg/Library/CompilerIntrinsicsLib/ArmCompilerIntrinsicsLib.inf [AARCH64]
INFO - Trim --asm-file -o /home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/Build/QemuSbsaPkg/DEBUG_GCC5/AARCH64/MdePkg/Library/CompilerIntrinsicsLib/ArmCompilerIntrinsicsLib/OUTPUT/AArch64/Atomics.i -i /home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/Build/QemuSbsaPkg/DEBUG_GCC5/AARCH64/MdePkg/Library/CompilerIntrinsicsLib/ArmCompilerIntrinsicsLib/OUTPUT/inc.lst /home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/MU_BASECORE/MdePkg/Library/CompilerIntrinsicsLib/AArch64/Atomics.S
INFO - Building ... /home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/MU_BASECORE/MdePkg/Library/UefiDriverEntryPoint/UefiDriverEntryPoint.inf [AARCH64]
INFO - "gcc" -MMD -MF /home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/Build/QemuSbsaPkg/DEBUG_GCC5/AARCH64/MdePkg/Library/UefiDriverEntryPoint/UefiDriverEntryPoint/OUTPUT/DriverEntryPoint.obj.deps  -g -Os -fshort-wchar -fno-builtin -fno-strict-aliasing -Wall -Werror -Wno-array-bounds -include AutoGen.h -fno-common -fstack-protector -ffunction-sections -fdata-sections -DSTRING_ARRAY_NAME=UefiDriverEntryPointStrings -g -Os -fshort-wchar -fno-builtin -fno-strict-aliasing -Wall -Werror -Wno-array-bounds -include AutoGen.h -fno-common -fstack-protector -mlittle-endian -fno-short-enums -fverbose-asm -funsigned-char -ffunction-sections -fdata-sections -Wno-address -fno-asynchronous-unwind-tables -fno-unwind-tables -fno-pic -fno-pie -ffixed-x18 -mstack-protector-guard=global -mcmodel=small -flto -Wno-unused-but-set-variable -Wno-unused-const-variable -DDISABLE_NEW_DEPRECATED_INTERFACES -c -o /home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/Build/QemuSbsaPkg/DEBUG_GCC5/AARCH64/MdePkg/Library/UefiDriverEntryPoint/UefiDriverEntryPoint/OUTPUT/./DriverEntryPoint.obj -I/home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/MU_BASECORE/MdePkg/Library/UefiDriverEntryPoint -I/home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/Build/QemuSbsaPkg/DEBUG_GCC5/AARCH64/MdePkg/Library/UefiDriverEntryPoint/UefiDriverEntryPoint/DEBUG -I/home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/MU_BASECORE/MdePkg -I/home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/MU_BASECORE/MdePkg/Include -I/home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/MU_BASECORE/MdePkg/Test/UnitTest/Include -I/home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/MU_BASECORE/MdePkg/Test/Mock/Include -I/home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/MU_BASECORE/MdePkg/Library/MipiSysTLib/mipisyst/library/include -I/home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/MU_BASECORE/MdePkg/Include/AArch64 /home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/MU_BASECORE/MdePkg/Library/UefiDriverEntryPoint/DriverEntryPoint.c
INFO - Building ... /home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/MU_BASECORE/MdeModulePkg/Library/BaseRngLibTimerLib/BaseRngLibTimerLib.inf [AARCH64]
INFO - gcc: error: unrecognized command-line option ‘-mlittle-endian’
INFO - make: *** [GNUmakefile:310: /home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/Build/QemuSbsaPkg/DEBUG_GCC5/AARCH64/MdePkg/Library/UefiDriverEntryPoint/UefiDriverEntryPoint/OUTPUT/DriverEntryPoint.obj] Error 1
INFO - "gcc" -MMD -MF /home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/Build/QemuSbsaPkg/DEBUG_GCC5/AARCH64/MdeModulePkg/Library/BaseRngLibTimerLib/BaseRngLibTimerLib/OUTPUT/RngLibTimer.obj.deps  -g -Os -fshort-wchar -fno-builtin -fno-strict-aliasing -Wall -Werror -Wno-array-bounds -include AutoGen.h -fno-common -fstack-protector -ffunction-sections -fdata-sections -DSTRING_ARRAY_NAME=BaseRngLibTimerLibStrings -g -Os -fshort-wchar -fno-builtin -fno-strict-aliasing -Wall -Werror -Wno-array-bounds -include AutoGen.h -fno-common -fstack-protector -mlittle-endian -fno-short-enums -fverbose-asm -funsigned-char -ffunction-sections -fdata-sections -Wno-address -fno-asynchronous-unwind-tables -fno-unwind-tables -fno-pic -fno-pie -ffixed-x18 -mstack-protector-guard=global -mcmodel=small -flto -Wno-unused-but-set-variable -Wno-unused-const-variable -DDISABLE_NEW_DEPRECATED_INTERFACES -mstrict-align -mgeneral-regs-only -c -o /home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/Build/QemuSbsaPkg/DEBUG_GCC5/AARCH64/MdeModulePkg/Library/BaseRngLibTimerLib/BaseRngLibTimerLib/OUTPUT/./RngLibTimer.obj -I/home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/MU_BASECORE/MdeModulePkg/Library/BaseRngLibTimerLib -I/home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/Build/QemuSbsaPkg/DEBUG_GCC5/AARCH64/MdeModulePkg/Library/BaseRngLibTimerLib/BaseRngLibTimerLib/DEBUG -I/home/edhcha01/MU/NEW_BUILD/mu_tiano_pl [... cut ...]
INFO - --------------Cmd Output Finished---------------
INFO - --------- Running Time (mm:ss): 00:13 ----------
INFO - ----------- Return Code: 0x00000001 ------------
INFO - ------------------------------------------------
ERROR - Compiler #7000 from : Failed to execute command
ERROR - Compiler #7000 from : Failed to execute command
ERROR - Compiler #7000 from : Failed to execute command
ERROR - Compiler #7000 from : Failed to execute command
ERROR - EDK2 #002 from : Failed to build module
CRITICAL - Build failed
```

edhay, 2025-06-05:
Thank you @apop5, Will try these steps and get back to you.

edhay, 2025-06-10:
Hi @apop5 

I followed the below steps:
```
source  ./.venv/bin/activate
export GCC5_AARCH64_PREFIX=aarch64-linux-gnu-
export PATH=/usr/bin:$PATH
```

(```
.venv) edhcha01@a076513:~/MU/NEW_BUILD/mu_tiano_platforms$ which aarch64-linux-gnu-gcc
/usr/bin/aarch64-linux-gnu-gcc
```

Compiler is found successfully

```
(.venv) edhcha01@a076513:~/MU/NEW_BUILD/mu_tiano_platforms$ ${GCC5_AARCH64_PREFIX}gcc
aarch64-linux-gnu-gcc: fatal error: no input files
compilation terminated.
```

However 

`stuart_build -c Platforms/QemuSbsaPkg/PlatformBuild.py TOOL_CHAIN_TAG=GCC5`

gives the error

```
INFO - Getting git ignore files...
INFO - Ignoring git ignore files...
INFO -   27 files are ignored by git
INFO -   28 files after removing ignored files
INFO -
Starting macro check on 28 files.
ERROR - Path for GCC5_AARCH64_PREFIX toolchain is invalid
CRITICAL - Failed in check aarch64
ERROR - Plugin Failed: Linux GCC5 Tool Chain Support returned -2
CRITICAL - Pre Build failed
PROGRESS - End time: 2025-06-11 01:58:19.546607  Total time Elapsed: 0:00:00
SECTION - Log file is located at: /home/edhcha01/MU/NEW_BUILD/mu_tiano_platforms/Build/BUILDLOG_QemuSbsaPkg.txt
SECTION - Summary
ERROR - Error
```

This is the usual method in cross compiling. Is there anything additional to be done for stuart_build?

apop5, 2025-06-12:
I believe the stuart build system isn't using the PATH: it is trying to no rely on the enviornment being setup correctly because it doesn't want to contaminate/ modify something in the system that may end up breaking other things you are attempting to build.
@Javagedes Do you want to make sure my statement about stuart is correct?

if you are able to change your export:
export GCC5_AARCH64_PREFIX=/usr/bin/aarch64-linux-gnu-

then you should be able to build correctly.

Javagedes, 2025-06-12:
Stuart (python in general actually), takes a "screenshot" of your environment variables, including PATH, when it starts. Stuart then mucks around with all of your different environment variables including PATH to set things up temporarily, without contaminating / permanently modifying the system, as @apop5 mentioned.

This is the exact commit and portion that is failing for you:

https://github.com/microsoft/mu_basecore/blob/dda9d4e1e359fa70aaae53ad37e5e63806de19b4/BaseTools/Plugin/LinuxGcc5ToolChain/LinuxGcc5ToolChain.py#L77-L99

I agree that setting `GCC5_AARCH64_PREFIX=aarch64-linux-gnu-` was incorrect, and it should have been set to `GCC5_AARCH64_PREFIX=/usr/bin/aarch64-linux-gnu-`, however it most scenarios, in a normal ubuntu image, you actually don't want to set those environment variables at all. One benefit of stuart is it does it for you (as seen in the link I provided)

apop5, 2025-06-20:
> i have run into the exact same series of issues as @edhay, starting from the nuget issue to the GCC issue
> 
> the GCC issue is fixed by running `export GCC5_AARCH64_PREFIX=/usr/bin/aarch64-linux-gnu-` (for some reason, stuart isn't doing it automatically).
> 
> i subsequently ran into a clang issue which was resolved by installing clang.
> 
> the error that i'm currently blocked on is it attempting and failing to compile standard headers
> ```
> INFO - [18/131] CC secure_qemu_aarch64_clang/obj/src/vm.vm.o
> INFO - FAILED: secure_qemu_aarch64_clang/obj/src/vm.vm.o
> INFO - clang -MMD -MF secure_qemu_aarch64_clang/obj/src/vm.vm.o.d     -DVM_TOOLCHAIN=0 -DGIC_VERSION=3 -DGICD_BASE=0x40060000 -DGICR_BASE=0x40080000 -DGICR_FRAMES=8 -DHEAP_PAGES=180 -DMAX_CPUS=8 -DMAX_VMS=16 -DLOG_LEVEL=LOG_LEVEL_VERBOSE -DENABLE_ASSERTIONS=1 -DPARTITION_MAX_UUIDS=4 -DPARTITION_MAX_MEMORY_REGIONS=8 -DPARTITION_MAX_DEVICE_REGIONS=8 -DPARTITION_MAX_DMA_DEVICES=2 -DPARTITION_MAX_INTERRUPTS_PER_DEVICE=4 -DPARTITION_MAX_STREAMS_PER_DEVICE=4 -DHF_NUM_INTIDS=1024 -DSECURE_WORLD=1 -I../../inc -I../../inc/vmapi -I../../src/arch/aarch64/inc -Isecure_qemu_aarch64_clang/gen/offset_size_header -ffunction-sections -fdata-sections -flto -fno-builtin -ffreestanding -fpic -mcpu=cortex-a57+nofp -mstrict-align  -mgeneral-regs-only -march=armv8.5-a -target aarch64-none-elf -fcolor-diagnostics -nostdinc -isystem /include -gdwarf-4 -O2 -Wall -Wextra -Wpedantic -Werror -Wshift-sign-overflow -fstack-protector-all -std=c11 -c ../../src/vm.c -o secure_qemu_aarch64_clang/obj/src/vm.vm.o
> INFO - In file included from ../../src/vm.c:9:
> INFO - ../../inc/hf/vm.h:11:10: fatal error: 'stdatomic.h' file not found
> INFO -    11 | #include <stdatomic.h>
> INFO -       |          ^~~~~~~~~~~~~
> INFO - 1 error generated.
> INFO - ninja: build stopped: subcommand failed.
> INFO - make: *** [Makefile:105: all] Error 1
> ```
> 
> i have not changed any code and the command i'm running is `stuart_build -c Platforms/QemuSbsaPkg/PlatformBuild.py TOOL_CHAIN_TAG=GCC5 BLD_*_PERF_TRACE_ENABLE=1 --FlashRom`

These errors are coming from hafnium's build system.

Try changing into the hafnium Su folder and running git clean -dfx and then rerunning the build.

apop5, 2025-06-24:
@edhay 

We have a PR up to mention the need for the export statement in the building document.

Thanks for reporting the issue.